gaoweifan/pyRadar #1: 请教一些关于IWR1843的问题

xiongsp, 2023-02-16:
我们团队正在做关于1843的项目，目前在github上没有找到相关IWR1843能处理出多普勒等数据的仓库。有以下几个问题还望赐教：
1. 不知道我们是否可以不需要DCA1000来完成对1843数据的读取、处理等。例如处理出1843识别到的x、y、z、速度、多普勒、角度等信息。
2. 如果DCA1000是必须的话，它具体的作用只是为了处理出速度、多普勒等信息吗？

万分感谢。

gaoweifan, 2023-02-16:
1. 可以的，DCA1000只是用于采集ADC采样的原始IQ数据，该数据通过LVDS高速接口传输，需要用到DCA1000将LVDS接口的数据打包成UDP数据包通过网口传输到电脑。如果不需要原始IQ数据可以利用IWR1843片内的Arm CPU和DSP来处理得到点云、多普勒图等数据，通过串口UART的方式接收数据，无需DCA1000的帮助。详情可以参考[mmwave SDK](https://www.ti.com/tool/MMWAVE-SDK)的demo，配合[mmWave_Demo_Visualizer](https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/3.6.0/)使用。另外，本仓库的代码也需要配合mmwave SDK的demo程序使用，无需修改demo代码，直接烧录已编译的固件。
2. DCA1000不是必须的，其作用是将LVDS接口的数据打包成UDP数据包通过网口传输到电脑，同时利用SPI对IWR1843芯片进行控制。利用SPI控制需要用到[mmwave studio](https://www.ti.com/tool/MMWAVE-STUDIO)，并且烧录studio附带的专用固件。而本仓库未使用SPI作为控制接口，本仓库利用UART作为控制接口对IWR1843芯片进行配置，固件则需使用mmwave SDK的demo程序。

closed 2023-02-18 by gaoweifan

Nobreakfast, 2023-08-21:
> * 可以的，DCA1000只是用于采集ADC采样的原始IQ数据，该数据通过LVDS高速接口传输，需要用到DCA1000将LVDS接口的数据打包成UDP数据包通过网口传输到电脑。如果不需要原始IQ数据可以利用IWR1843片内的Arm CPU和DSP来处理得到点云、多普勒图等数据，通过串口UART的方式接收数据，无需DCA1000的帮助。详情可以参考[mmwave SDK](https://www.ti.com/tool/MMWAVE-SDK)的demo，配合[mmWave_Demo_Visualizer](https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/3.6.0/)使用。另外，本仓库的代码也需要配合mmwave SDK的demo程序使用，无需修改demo代码，直接烧录已编译的固件。
> * DCA1000不是必须的，其作用是将LVDS接口的数据打包成UDP数据包通过网口传输到电脑，同时利用SPI对IWR1843芯片进行控制。利用SPI控制需要用到[mmwave studio](https://www.ti.com/tool/MMWAVE-STUDIO)，并且烧录studio附带的专用固件。而本仓库未使用SPI作为控制接口，本仓库利用UART作为控制接口对IWR1843芯片进行配置，固件则需使用mmwave SDK的demo程序。

大佬您好，我想问一下，如果给1843编程，可以直接从串口读取adc数据吗？

Nobreakfast, 2023-08-22:
> @Nobreakfast 可以的，但是串口很慢，一帧会传很久，而且如果一帧数据量太大内存还会不足存不下，还需要自己写相关的串口传输程序，所以还是推荐用lvds传输原始数据

感谢回复。