gaoweifan/pyRadar #2: LVDS_PATH_ERR LED lighting.

wantine, 2023-05-26:
Hello. I burned xwr18xx_mmw_demo and set the adcbufCfg to -1 0 1 1 1 and lvdsStreamCfg to -1 0 1 0. And the SOP mode of 1843 is 001 (SOP0: on, SOP1: off, SOP2: off). But when I run captureall.py, the program stopped abnormally. Nothing is stored and the LED called LVDS_PATH_ERR is lighting. I wonder if I missed some hardware settings, or the SOP mode is wrong. Thank you.

gaoweifan, 2023-05-26:
Please send me the output of the program so I can identify the source of the problem more easily. Also, the cfg file from this repository might not work properly. You should use [mmWave_Demo_Visualizer](https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/3.6.0/) and `testParam.ipynb` to create and debug a cfg file and make sure it works. The `LVDS_PATH_ERR` red light means that LVDS has not received any data. This could be because the radar failed to start due to an incorrect configuration file.

wantine, 2023-05-26:
```
wait for reset
>>> sensorStop
b'sensorStop\n'
>>> flushCfg
b'flushCfg\n'
>>> dfeDataOutputMode 1
b'dfeDataOutputMode 1\n'
>>> channelCfg 15 7 0
b'channelCfg 15 7 0\n'
>>> adcCfg 2 1
b'adcCfg 2 1\n'
>>> adcbufCfg -1 0 1 1 1
b'adcbufCfg -1 0 1 1 1\n'
>>> profileCfg 0 77 974 7 40 0 0 100 1 64 2000 0 0 30
b'profileCfg 0 77 974 7 40 0 0 100 1 64 2000 0 0 30\n'
>>> chirpCfg 0 0 0 0 0 0 0 1
b'chirpCfg 0 0 0 0 0 0 0 1\n'
>>> chirpCfg 1 1 0 0 0 0 0 4
b'chirpCfg 1 1 0 0 0 0 0 4\n'
>>> chirpCfg 2 2 0 0 0 0 0 2
b'chirpCfg 2 2 0 0 0 0 0 2\n'
>>> frameCfg 0 2 32 0 200 1 0
b'frameCfg 0 2 32 0 200 1 0\n'
>>> lowPower 0 0
b'lowPower 0 0\n'
>>> guiMonitor -1 1 1 1 1 1 1
b'guiMonitor -1 1 1 1 1 1 1\n'
>>> cfarCfg -1 0 2 8 4 3 0 15 1
b'cfarCfg -1 0 2 8 4 3 0 15 1\n'
>>> cfarCfg -1 1 0 8 4 4 1 15 1
b'cfarCfg -1 1 0 8 4 4 1 15 1\n'
>>> multiObjBeamForming -1 1 0.5
b'multiObjBeamForming -1 1 0.5\n'
>>> clutterRemoval -1 0
b'clutterRemoval -1 0\n'
>>> calibDcRangeSig -1 0 -5 8 256
b'calibDcRangeSig -1 0 -5 8 256\n'
>>> extendedMaxVelocity -1 0
b'extendedMaxVelocity -1 0\n'
>>> lvdsStreamCfg -1 0 1 0
b'lvdsStreamCfg -1 0 1 0\n'
>>> compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0
b'compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0\n'
>>> measureRangeBiasAndRxChanPhase 0 1.5 0.2
b'measureRangeBiasAndRxChanPhase 0 1.5 0.2\n'
>>> CQRxSatMonitor 0 3 4 99 0
b'CQRxSatMonitor 0 3 4 99 0\n'
>>> CQSigImgMonitor 0 31 4
data reading process created
{'startFreq': 77, 'idleTime': 974.0, 'adc_valid_start_time': 7.0, 'rampEndTime': 40.0, 'freq_slope': 100.0, 'txStartTime': 1.0, 'samples': 64, 'sample_rate': 2000, 'chirps': 32, 'frame_periodicity': 200.0, 'rx': 4, 'tx': 3, 'IQ': 2, 'bytes': 2}
{'platfrom': '18', 'rxAntMask': 15, 'txAntMask': 7, 'dataSize': 2, 'dataType': 1, 'chirpMode': 1, 'chirpStartIdx': 0, 'chirpEndIdx': 2, 'numSubframes': 1, 'numRxChan': 4, 'numTxChan': 3, 'datacard_dataLoggingMode': 'raw'}
LVDSDataSizePerChirp:1280 must <= maxSendBytesPerChirp:152100
System connection check: success
FPGA Version : 2.8 [Record]
Config fpga: success
Config record packet delay: success
press ENTER to start capture...
data reading process started
Start streaming
>>> sensorStart
b'sensorStart\n'
min Packet Num: 676
```
Thank you so much. Sometime the output is like this. But most of the time it stopped at this sentence: ">>>sensorStart" and then the program seems stopped. And I wonder if I should keep the settings like the DCA1000EVM CLI Software User Guide. And when I use mmWaveStudio to capture raw data, SOP is 011, when I use Demo Visualizer the SOP is 001. Should I switch to 011 after creating the cfg file or keep it 001.

gaoweifan, 2023-05-26:
It appears that both the IWR1843 and DCA1000 are functioning properly, but the radar is not starting up correctly according to the preset configuration. This is usually due to errors in the cfg file. For example, if the inter-frame gap time is shorter than the time required to send a data packet, the previous frame's data packet may not be fully sent before the next frame starts transmitting, causing the radar to stop working. The solution is to replace the cfg file with one that has a lower data transmission volume and a larger inter-frame gap, such as reducing the number of sampling points or increasing the frame period. Generally, if the mmWave_Demo_Visualizer can run successfully, the configuration file should work fine in this program as well. You can try using mmWave_Demo_Visualizer to generate a validated configuration file directly. Additionally, the SOP should be set to "001" (functional mode) for normal operation.

wantine, 2023-05-26:
非常感谢。 我使用visualizer 导出profile 文件后 使用testParam.ipynb 检查，log信息为青色。 
[image]
我分别尝试 使用 此仓库的profile文件和自己导出的文件，对于这两个配置文件目前的运行情况是一样的：LVDS_PATH_ERR不亮红灯了，并且DATA_TRAN_PRG也会闪烁，似乎已经正确触发了frame，IWR1843板子上GPIO_2黄灯亮起。
[image]
CaptureAll.py 的 Output信息如下：
`(1843) C:\Users\mmwavevitalsign\Radar>C:/Users/aiot/anaconda3/envs/1843/python.exe c:/Users/mmwavevitalsign/Radar/captureAll.py
wait for reset
>>> sensorStop
b'sensorStop\nIgnored: Sensor is already stopped\n\rDone\n\rmmwDemo:/>'
>>> flushCfg
b'flushCfg\nDone\n\rmmwDemo:/>'
>>> dfeDataOutputMode 1
b'dfeDataOutputMode 1\nDone\n\rmmwDemo:/>'
>>> channelCfg 15 7 0
b'channelCfg 15 7 0\nDone\n\rmmwDemo:/>'
>>> adcCfg 2 1
b'adcCfg 2 1\nDone\n\rmmwDemo:/>'
>>> adcbufCfg -1 0 1 1 1
b'adcbufCfg -1 0 1 1 1\nDone\n\rmmwDemo:/>'
>>> profileCfg 0 77 974 7 40 0 0 100 1 64 2000 0 0 30
b'profileCfg 0 77 974 7 40 0 0 100 1 64 2000 0 0 30\nDone\n\rmmwDemo:/>'
>>> chirpCfg 0 0 0 0 0 0 0 1
b'chirpCfg 0 0 0 0 0 0 0 1\nDone\n\rmmwDemo:/>'
>>> chirpCfg 1 1 0 0 0 0 0 4
b'chirpCfg 1 1 0 0 0 0 0 4\nDone\n\rmmwDemo:/>'
>>> chirpCfg 2 2 0 0 0 0 0 2
b'chirpCfg 2 2 0 0 0 0 0 2\nDone\n\rmmwDemo:/>'
>>> frameCfg 0 2 32 0 200 1 0
b'frameCfg 0 2 32 0 200 1 0\nDone\n\rmmwDemo:/>'
>>> lowPower 0 0
b'lowPower 0 0\nDone\n\rmmwDemo:/>'
>>> guiMonitor -1 1 1 1 1 1 1
b'guiMonitor -1 1 1 1 1 1 1\nDone\n\rmmwDemo:/>'
>>> cfarCfg -1 0 2 8 4 3 0 15 1
b'cfarCfg -1 0 2 8 4 3 0 15 1\nDone\n\rmmwDemo:/>'
>>> cfarCfg -1 1 0 8 4 4 1 15 1
b'cfarCfg -1 1 0 8 4 4 1 15 1\nDone\n\rmmwDemo:/>'
>>> multiObjBeamForming -1 1 0.5
b'multiObjBeamForming -1 1 0.5\nDone\n\rmmwDemo:/>'
>>> clutterRemoval -1 0
b'clutterRemoval -1 0\nDone\n\rmmwDemo:/>'
>>> calibDcRangeSig -1 0 -5 8 256
b'calibDcRangeSig -1 0 -5 8 256\nDone\n\rmmwDemo:/>'
>>> extendedMaxVelocity -1 0
b'extendedMaxVelocity -1 0\nDone\n\rmmwDemo:/>'
>>> lvdsStreamCfg -1 0 1 0
b'lvdsStreamCfg -1 0 1 0\nDone\n\rmmwDemo:/>'
>>> compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0
b'compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0\nDone\n\rmmwDemo:/>'
>>> measureRangeBiasAndRxChanPhase 0 1.5 0.2
b'measureRangeBiasAndRxChanPhase 0 1.5 0.2\nDone\n\rmmwDemo:/>'
>>> CQRxSatMonitor 0 3 4 99 0
b'CQRxSatMonitor 0 3 4 99 0\nDone\n\rmmwDemo:/>'
>>> CQSigImgMonitor 0 31 4
b'CQSigImgMonitor 0 31 4\nDone\n\rmmwDemo:/>'
>>> analogMonitor 0 0
b'analogMonitor 0 0\nDone\n\rmmwDemo:/>'
>>> aoaFovCfg -1 -90 90 -90 90
FPGA Version : 2.8 [Record]
Config fpga: success
Config record packet delay: success
press ENTER to start capture...
data reading process started
Start streaming
>>> sensorStart

(1843) C:\Users\mmwavevitalsign\Radar>`

程序没有 输出 time elapsed(s) 及之后，如save file的信息。

gaoweifan, 2023-05-26:
如果已经触发了frame却没有后续输出的话很可能是UDP数据未完整采集或采集代码不兼容的问题。
针对前者，请检查cf.json文件中的"lvdsMode"是否已改为2，即2通道LVDS模式，IWR1843仅支持2通道传输，具体请参阅相关官方文档。
针对后者，可以尝试切换到【方法二】，即注释掉第89行与第97行代码，并取消第98行代码的注释。
```python
# numframes_out,sortInC_out = dca.fastRead_in_Cpp_async_start(numframes,sortInC=True)
# data_buf = dca.fastRead_in_Cpp_async_wait(numframes_out,sortInC_out) # 【方法一】等待异步线程结束
data_buf = dca.fastRead_in_Cpp(numframes,sortInC=True) # 【方法二】同步调用(会丢失开始采集前的包)
``` 
若还是不行，则可以设置`sortInC=False`利用python进行数据包的排序。本代码部分特性需要用到支持C++17及以上的编译器，若编译器过于古老则很有可能出现各种兼容性问题。

If you have triggered the frame but there is no subsequent output, it is likely due to incomplete UDP data acquisition or compatibility issues with the acquisition code.

For the former issue, please check if the "lvdsMode" in the cf.json file has been changed to 2, which corresponds to the 2-lane LVDS mode. The IWR1843 only supports 2-lane transmission. Please refer to the relevant official documentation for more details.

For the latter issue, you can try switching to "Method 2" by commenting out line 89 and line 97, and uncommenting line 98.
```python
# numframes_out,sortInC_out = dca.fastRead_in_Cpp_async_start(numframes,sortInC=True)
# data_buf = dca.fastRead_in_Cpp_async_wait(numframes_out,sortInC_out) # [Method 1] Waiting for asynchronous threads to finish.
data_buf = dca.fastRead_in_Cpp(numframes,sortInC=True) # 
[Method 2] Synchronous Invocation (May result in the loss of packets before starting the acquisition)
``` 
If the issue persists, you can set `sortInC=False` to use Python for packet sorting. Some features of this code require a compiler that supports C++17 or higher. If the compiler is too old, it is likely to encounter various compatibility issues.

wantine, 2023-05-27:
Wow, awesome. I switched to "Method 2" and it seems work properly. Thank you so much. 
[image]

closed 2023-05-27 by gaoweifan