gaoweifan/pyRadar #3: 循环采集

Yinsongxu, 2023-07-12:
我在循环采集的时候，第一个循环没有问题，从第二个开始，丢包率就会变成接近100%，请问这是什么情况呢？

gaoweifan, 2023-07-12:
请贴上你循环采集的代码

Yinsongxu, 2023-07-12:
```python
import traceback
import time
from mmwave.dataloader import DCA1000
from mmwave.dataloader.radars import TI
import numpy as np
dca = None
radar = None

try:
    dca = DCA1000()

    dca.reset_radar()
    dca.reset_fpga()
    print("wait for reset")
    time.sleep(1)

    dca_config_file = "configFiles\cf.json" 
    radar_config_file = r"configFiles\xWR1843_profile_3D.cfg" 
    numframes=10
    radar = TI(cli_loc='COM6', data_loc='COM5',data_baud=921600,config_file=radar_config_file,verbose=True)
    radar.setFrameCfg(numframes)
    dca.configure(dca_config_file,radar_config_file)  

    for i in range(5):
        radar.start_read_process()
        dca.stream_start()
        radar.startSensor()

        data_buf = dca.fastRead_in_Cpp(numframes,sortInC=True) 
        filename=f"raw_data{i}.bin"
        data_buf.tofile(filename)

        radar.stopSensor()
        radar.stop_read_process()
        time.sleep(0.01)
        
except Exception as e:
    traceback.print_exc()
finally:
    if dca is not None:
        dca.close()
    if radar is not None:
        radar.cli_port.close()

```
修改numframes时，丢包率会变化
> wait for reset
>>> sensorStop
b'sensorStop\nIgnored: Sensor is already stopped\n\rDone\n\rmmwDemo:/>'
>>> flushCfg
b'flushCfg\nDone\n\rmmwDemo:/>'
>>> dfeDataOutputMode 1
b'dfeDataOutputMode 1\nDone\n\rmmwDemo:/>'
>>> channelCfg 15 7 0
b'channelCfg 15 7 0\nDone\n\rmmwDemo:/>'
>>> adcCfg 2 1
b'adcCfg 2 1\nDone\n\rmmwDemo:/>'
>>> adcbufCfg -1 0 1 1 1
b'adcbufCfg -1 0 1 1 1\nDone\n\rmmwDemo:/>'
>>> profileCfg 0 77 20 7 40 0 0 100 1 64 2000 0 0 30
b'profileCfg 0 77 20 7 40 0 0 100 1 64 2000 0 0 30\nDone\n\rmmwDemo:/>'
>>> chirpCfg 0 0 0 0 0 0 0 1
b'chirpCfg 0 0 0 0 0 0 0 1\nDone\n\rmmwDemo:/>'
>>> chirpCfg 1 1 0 0 0 0 0 4
b'chirpCfg 1 1 0 0 0 0 0 4\nDone\n\rmmwDemo:/>'
>>> chirpCfg 2 2 0 0 0 0 0 2
b'chirpCfg 2 2 0 0 0 0 0 2\nDone\n\rmmwDemo:/>'
>>> frameCfg 0 2 32 0 100 1 0
b'frameCfg 0 2 32 0 100 1 0\nDone\n\rmmwDemo:/>'
>>> lowPower 0 0
b'lowPower 0 0\nDone\n\rmmwDemo:/>'
>>> guiMonitor -1 1 1 1 1 1 1
b'guiMonitor -1 1 1 1 1 1 1\nDone\n\rmmwDemo:/>'
>>> cfarCfg -1 0 2 8 4 3 0 15 1
b'cfarCfg -1 0 2 8 4 3 0 15 1\nDone\n\rmmwDemo:/>'
>>> cfarCfg -1 1 0 8 4 4 1 15 1
b'cfarCfg -1 1 0 8 4 4 1 15 1\nDone\n\rmmwDemo:/>'
>>> multiObjBeamForming -1 1 0.5
b'multiObjBeamForming -1 1 0.5\nDone\n\rmmwDemo:/>'
>>> clutterRemoval -1 0
b'clutterRemoval -1 0\nDone\n\rmmwDemo:/>'
>>> calibDcRangeSig -1 0 -5 8 256
b'calibDcRangeSig -1 0 -5 8 256\nDone\n\rmmwDemo:/>'
>>> extendedMaxVelocity -1 0
b'extendedMaxVelocity -1 0\nDone\n\rmmwDemo:/>'
>>> lvdsStreamCfg -1 0 1 0
b'lvdsStreamCfg -1 0 1 0\nDone\n\rmmwDemo:/>'
>>> compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0
b'compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0\nDone\n\rmmwDemo:/>'
>>> measureRangeBiasAndRxChanPhase 0 1.5 0.2
b'measureRangeBiasAndRxChanPhase 0 1.5 0.2\nDone\n\rmmwDemo:/>'
>>> CQRxSatMonitor 0 3 4 99 0
b'CQRxSatMonitor 0 3 4 99 0\nDone\n\rmmwDemo:/>'
>>> CQSigImgMonitor 0 31 4
b'CQSigImgMonitor 0 31 4\nDone\n\rmmwDemo:/>'
>>> analogMonitor 0 0
b'analogMonitor 0 0\nDone\n\rmmwDemo:/>'
>>> aoaFovCfg -1 -90 90 -90 90
b'aoaFovCfg -1 -90 90 -90 90\nDone\n\rmmwDemo:/>'
>>> cfarFovCfg -1 0 0 2.40
b'cfarFovCfg -1 0 0 2.40\nDone\n\rmmwDemo:/>'
>>> cfarFovCfg -1 1 -0.32 0.32
b'cfarFovCfg -1 1 -0.32 0.32\nDone\n\rmmwDemo:/>'
>>> calibData 0 0 0
b'calibData 0 0 0\nDone\n\rmmwDemo:/>'
>>> frameCfg 0 2 32 10 100.0 1 0.0
b'frameCfg 0 2 32 10 100.0 1 0.0\nDone\n\rmmwDemo:/>'
{'startFreq': 77, 'idleTime': 20.0, 'adc_valid_start_time': 7.0, 'rampEndTime': 40.0, 'freq_slope': 100.0, 'txStartTime': 1.0, 'samples': 64, 'sample_rate': 2000, 'chirps': 32, 'frame_periodicity': 100.0, 'rx': 4, 'tx': 3, 'IQ': 2, 'bytes': 2}
{'platfrom': '18', 'rxAntMask': 15, 'txAntMask': 7, 'dataSize': 2, 'dataType': 1, 'chirpMode': 1, 'chirpStartIdx': 0, 'chirpEndIdx': 2, 'numSubframes': 1, 'numRxChan': 4, 'numTxChan': 3, 'datacard_dataLoggingMode': 'raw'}
LVDSDataSizePerChirp:1280 must <= maxSendBytesPerChirp:9000
System connection check: success
FPGA Version : 2.8 [Record]
Config fpga: success
Config record packet delay: success
data reading process started
Start streaming
>>> sensorStart
b'sensorStart\n'
min Packet Num: 676
first Packet Num:1,last Packet Num:676
received packet num:676,expected packet num:676,loss:0.00%
>>> sensorStop
b'Debug: Init Calibration Status = 0x1ffe\n\rDone\n\rmmwDemo:/>sensorStop\nDone\n\rmmwDemo:/>'
data reading process exit
data reading process started
receive upd packet error
>>> sensorStart 0
b'sensorStart 0\nDone\n\rmmwDemo:/>'
min Packet Num: 676
[fpga_udp]udp time out
first Packet Num:811,last Packet Num:0
received packet num:542,expected packet num:676,loss:19.82%
>>> sensorStop
b'sensorStop\nDone\n\rmmwDemo:/>'
data reading process exit
data reading process started
Start streaming
>>> sensorStart 0
b'sensorStart 0\nDone\n\rmmwDemo:/>'
min Packet Num: 676
[fpga_udp]udp time out
first Packet Num:1486,last Packet Num:0
received packet num:543,expected packet num:676,loss:19.67%
>>> sensorStop
b'sensorStop\nDone\n\rmmwDemo:/>'
data reading process exit
data reading process started
receive upd packet error
>>> sensorStart 0
b'sensorStart 0\nDone\n\rmmwDemo:/>'
min Packet Num: 676
[fpga_udp]udp time out
first Packet Num:2161,last Packet Num:0
received packet num:544,expected packet num:676,loss:19.53%
>>> sensorStop
b'sensorStop\nDone\n\rmmwDemo:/>'
data reading process exit
data reading process started
Start streaming
>>> sensorStart 0
b'sensorStart 0\nDone\n\rmmwDemo:/>'
min Packet Num: 676
[fpga_udp]udp time out
first Packet Num:2836,last Packet Num:0
received packet num:531,expected packet num:676,loss:21.45%
>>> sensorStop
b'sensorStop\nDone\n\rmmwDemo:/>'
data reading process exit
Close socket: success
Stop sensor, Close com port

gaoweifan, 2023-07-12:
### 问题产生原因
一帧数据大小不一定刚好等于整数个UDP数据包大小，可能一帧数据需要10.3个数据包才能传输完，那就必须采集11个数据包才能解析出一帧的数据。而那多余的0.7个数据包的内容则是下一帧的数据，在采集时被丢弃了，下一帧内容就不完整，必须从下下帧数据开始采集。**而dca.fastRead_in_Cpp()函数不会检查到达的数据包是否是一帧的第一个数据包**，不从一帧的开头开始采集的话，很有可能就会导致从一帧的中间开始采集，采集到这一帧的4个数据包和下一帧的7个数据包。在解析的时候就会发现只有4个数据包是有效数据，丢掉了多余的7个数据包，从而产生大量的丢包现象。而你的代码每次都重新启动雷达从头开始采集，看起来不会出现这个问题，但实际上SOCKET是有缓冲区的，如果**缓冲区内的数据没有清空**，同样会造成上述问题。
### 解决方法
**采用dca.fastRead_in_Cpp_thread_get()函数进行采集**，该函数考虑到了从中间开始采集的问题，会等待下一帧的第一个数据包到来时开始采集。具体使用方法及循环采集示例可以**参考realTimeProc_AWR2243.py**，按照里面的步骤来绝对没问题。需要注意的是，使用该方法循环采集无需每次都重新启动雷达，雷达及DCA采集板只需启动一次，并且设置帧数为0，无限采集下去，通过发送命令来控制是否停止采集。所以理论上来说只需要长这样就行：
```python
# 这里省略掉了重启、配置文件等初始代码
numLoops=50
radar.setFrameCfg(0)  # numframes==0，循环采集
dca.stream_start() # 启动DCA
dca.fastRead_in_Cpp_thread_start(frameNumInBuf) # 启动udp采集线程
radar.startSensor() # 启动雷达

for i in range(numLoops):
    data_buf = dca.fastRead_in_Cpp_thread_get(numframes,verbose=True,sortInC=True)  # 数据采集
    postProc(data_buf)  # 数据处理或保存

radar.stopSensor() # 停止雷达
dca.fastRead_in_Cpp_thread_stop() # 停止udp采集线程(必须先于stream_stop调用，即UDP接收时不能同时发送)
dca.stream_stop()  # DCA停止采集
```

closed 2023-07-13 by Yinsongxu