gaoweifan/pyRadar #6: module 'fpga_udp' has no attribute 'radar_start_read_thread'

fiala10, 2023-07-24:
Hi, 
I would like to use your code for my laboratory test, but I am stuck on this error.

module 'fpga_udp' has no attribute 'radar_start_read_thread'

Do you have any ideas on how to solve this issue?

Thanks a lot. 

Martin

[image]

gaoweifan, 2023-07-25:
Have you installed 'fpga_udp' module as required? Check if you did by running `pip show --files fpga_udp`.

fiala10, 2023-07-28:
Thank you, now it is working. However, I stucked again, when i start command python captureAll.py

I got his message...FPGA Version : 2.8 [Record]
Config fpga: success
Config record packet delay: success
press ENTER to start capture...
data reading process started
Start streaming
**>>> sensorStart
b''
min Packet Num: 676**

gaoweifan, 2023-07-28:
This issue has been raised before, please refer to #2 for more information, especially try switching to ["Method 2"](https://github.com/gaoweifan/pyRadar/issues/2#issuecomment-1564261036).

fiala10, 2023-07-28:
Thank you very much. Otherwise, do you have any idea, why I had recieved empty file? 
[image]

gaoweifan, 2023-07-28:
Have you tried [mmWave_Demo_Visualizer](https://dev.ti.com/gallery/view/mmwave/mmWave_Demo_Visualizer/ver/3.6.0/)? if the mmWave_Demo_Visualizer can run successfully, the configuration file should work fine in this program as well.

fiala10, 2023-07-28:
No, I have problem with connection DATA_OUTPUT_PORT. Do you think, that it would be problem in my drivers? 
[image]
[image]

gaoweifan, 2023-07-28:
What version of firmware are you using? I think  it is more likely that you’re using a firmware without UART data output feature. Pls install the firmware called “xwr18xx_mmw_demo” by following [mmwaveSDK](https://www.ti.com/tool/MMWAVE-SDK) user guide.

closed 2023-09-13 by gaoweifan