gaoweifan/pyRadar #7: The lights are LVDS_PATH_ERR when running with IWR6843ISK-ODS

ducouu, 2023-09-05:
Hello, the radar board I use is iwr6843isk-ods, flashed out_of_box_demo.bin, the sop of the radar board is set to functional mode 00110, has used testParam.ipynb and visualizer to check the cfg file, the log information is cyan, as shown in the figure below;
[image]

 when running capture all, lvds_path_err red light, the running result is as follows, Method 1 and Method 2 are used respectively, I want to know what documents need to be replaced for the change of radar board model, trouble, thanks
[image]
[image]

gaoweifan, 2023-09-08:
testParam.ipynb is designed to check parameters for IWR1843, it might not work for IWR6843, you should alter some restrains in testParam.ipynb based on the hardware feature of IWR6843. But as long as the params worked in mmWave_Demo_Visualizer, it should work in this program too. Although this program is designed for IWR1843/AWR2243, I believe it should work with IWR6843 as well.

Here's couple of things you should check:
1. have you tried to capture raw data using **mmwave studio**? if so, it means your hardware connection is solid, the problem probably comes from software or some **onboard switch** settings.
2. have you flashed right firmware? you should use firmware from mmwave SDK 3.6.0 located in "ti/mmwave_sdk_03_06_00_00-LTS/packages/ti/demo/xwr68xx/mmw/**xwr68xx_mmw_demo.bin**" as instructed in mmwave_sdk_user_guide.pdf.
3. have you configured the profile correctly? The LVDS_PATH_ERR red light means that LVDS has not received any data. So, changing the method of receiving won't work. This could be because the radar failed to start, or because there is no data coming from LVDS at all. This is usually caused by an incorrect configuration profile. Default profile in Visualizer disables the LVDS streaming. To enable it, please set the appropriate enable bits in profile, i.e., set the 3rd param of **lvdsStreamCfg** to 1.

Let me know if you have further questions.

ducouu, 2023-09-12:
Hello, I modified the configuration file and found that the data collection was successful! But now I have encountered a new problem. 
At present, I have transplanted the program to the ubuntu environment. When installing fpga_udp, the following problem occurred. How to solve it?
[image]
[image]

gaoweifan, 2023-09-12:
It says Python.h cannot be found, which is usually caused by missing development tools, header files and static library for building python modules. You can install it by `sudo apt install python3-dev` on ubuntu.

ducouu, 2023-09-13:
I ran successfully under linux system as you said and the method on the post, thank you very much for your help!

gaoweifan, 2023-09-13:
glad to hear that! i’m gonna close this issue, feel free to open a new issue if you have any other questions.

closed 2023-09-13 by gaoweifan