gaoweifan/pyRadar #8: Child processes affect the main process

ducouu, 2023-09-15:
Hello, currently I am porting the code to the linux system, I need to run the code you provided on the child process, other code runs in the main process, but now the child process affects the main process, causing the main process to freeze, I guess there is a problem in the following location, how to solve this problem?
[image]

gaoweifan, 2023-09-15:
dca.**fastRead_in_Cpp()** will block until all packets have been received. If you want to do something else while receiving, you can use dca.**fastRead_in_Cpp_thread_xxx()** ("xxx" can be "start", "get" and "stop") instead.
e.g.
```python
dca.stream_start()
dca.fastRead_in_Cpp_thread_start(frameNumInBuf)

# do whatever you want here

data_buf = dca.fastRead_in_Cpp_thread_get(numframes,verbose=True,sortInC=True)
postProc(data_buf)

dca.fastRead_in_Cpp_thread_stop()
dca.stream_stop()
dca.close()
```
see [realTimeProc.py](/gaoweifan/pyRadar/blob/main/realTimeProc.py) for more information.

ducouu, 2023-09-19:
Hello, I ported the program to the Jetson NX board of the arm processor under the Ubuntu system, and when I looked for the definition of the "udp_read_thread_get_frames" function, I found that I could not jump, which is shown in the following figure,
[image]
but when the mouse moves on the statement, you can see the function definition, is it because the library of the fpga_udp was not successfully installed? 
At the same time, during the running of the program, the result of [fpga_udp]udp time out will appear, what is the reason for this result?
[image]

gaoweifan, 2023-09-19:
1. fpga_udp library is written in CPP not python, as a result, any function inside fpga_udp such as `udp_read_thread_get_frames()` will not be visible in python project. If you want to dig into the function inside fpga_udp, just open its source file under fpga_udp subdirectory.
2. `[fpga_udp]udp time out` occurs when received less packets than expected. It indicates that there are some packets lost during transmission, which often happens when CPU performance is not enough to handle that amount of data. You can try to cut down fps or samples per chirp etc. to reduce data volume.

ducouu, 2023-09-22:
Hello, I put the function `dca.fastRead_in_Cpp_thread_get()` in the pyqt sub-process to run it in the hope that the sub-process will not conflict with the main process (UI interface). However, when the following statement in the function is executed in the sub-process, `recvData = fpga_udp.udp_read_thread_get_frames()`, causing the main interface to become unresponsive and stuck. How to solve this problem so that the two processes do not conflict?

gaoweifan, 2023-09-22:
你是在Jetson NX上运行的？根本就没有冲突这个说法，正常情况下两个线程不共享资源的话是没有任何交集的，不会说一个线程阻塞另一个线程，而你觉得有冲突大概率上因为Jetson NX的CPU太弱了，根本处理不过来大量的原始数据，导致接收一个完整的包所需的时间远大于一个完整的包被发送所消耗的时间，而dca.fastRead_in_Cpp_thread_get()会等待至少收到一个完整的包才会返回，你觉得卡住很可能就是因为这玩意儿收得太慢了，解决方案就是别用Jetson NX，至少别用Jetson NX运行QT,你只用Jetson NX进行采集+数据处理，然后回传数据到上位机进行显示不就好了，卡不是因为两个线程有什么冲突，根本就没这个说法，卡是因为你CPU太弱执行起来费劲，为了等待接收完毕才卡，这个与你编写多线程/多进程的技巧以及你运行的环境有关，与我这个库没有任何关系

closed 2023-11-08 by gaoweifan