gaoweifan/pyRadar #10: 关于用IWR1843+DCA1000实时抓取数据并处理时遇到的问题。

zhoXR, 2023-11-20:
我在抓包的时候发现，每一帧和下一帧之间会丢失68个包，原因应该是第一帧的最后一个数据包中可能有第二帧的开头，导致第二帧开头找不到，直到找到第三帧时开始，所以第二帧的数据就丢失了，不知道我的理解对不对，如果正确的话，应该怎么修复代码？
还有一个问题是我发现抓包的延迟有点高，如果想优化的话，应该从哪里入手呢？
[image]

gaoweifan, 2023-11-20:
如果你用的dca.**fastRead_in_Cpp_thread_xxx()** ("xxx" can be "start", "get" and "stop") ，那么是已经考虑到了你说的那种情况的。丢帧的原因大概是因为后处理速度慢于数据到达的速度，缓冲区大小是可以自定义的，即`frameNumInBuf`，如果数据没有及时取走数据，缓冲区满了就会导致未被取走的数据丢失，以腾出空间给新到来的数据。建议加大缓冲区并优化后处理程序，使其耗时小于捕获一帧数据的时间。

zhoXR, 2023-11-24:
感谢您的解答，根据您描述的方法，我仔细检查并调试后发现仍然存在问题：
由于我想测试抓包的过程，所以没有对抓到的数据做后处理，所计算的时间只是抓包的时间。我的缓冲区设的是16，而每次抓取1帧，应该问题不出在此处。
我又做了测试，假如我把每次抓取的帧数设成n=2~16，结果发现，他们都会在抓取n帧之后，丢失一帧，如果是我上面提到的那种情况的话，应该如何修改代码啊

gaoweifan, 2023-11-27:
好像当时确实是这么设计的，可以看看fpga_udp的C++代码[main.cpp](/gaoweifan/pyRadar/blob/main/fpga_udp/src/main.cpp)中的[udp_read_thread_get_frames](/gaoweifan/pyRadar/blob/main/fpga_udp/src/main.cpp#L296)函数，并作出相应的修改。

gaoweifan, 2023-12-08:
我没记错的话当时应该是这样设计的：
1. 有一个udp数据包接收队列，在一个udp接收线程里面每次从socket接收固定字节数据包（dca1000每次会发固定字节的udp数据过来，如果雷达最后一帧数据不足这么多字节则会补零，该数据包的结构可以看官方手册）存入该队列。
2. 调用[udp_read_thread_get_frames](https://github.com/gaoweifan/pyRadar/blob/main/fpga_udp/src/main.cpp#L296)时会从该队列挨个取数据包出来看数据包的sequence number，如果刚好是一帧的开头所在的数据包，则会从该包开始，取指定帧数所需要的所有数据包出来，否则就会丢掉该包并检查下一个数据包的sequence number。

显然问题就在这里，如果要改的话，我推荐在udp接收线程里，取到数据包后不要直接存入队列，而是取payload放入一个连续的缓冲区，这个缓冲区刚好是一帧的大小的整数倍，等到达数据包填满这个缓冲区后再将一整帧存入队列，并将多余的数据（下一帧的开头）存入新的缓冲区的开头。这样你就能从队列里取出一整帧数据，而不会导致丢掉一部分下一帧开头的数据导致这帧无法利用被丢弃。

Liwenlong524998, 2024-03-03:
您好，可以分享一下您的抓包程序吗？

Liwenlong524998, 2024-03-03:
您好，可以分享一下您的抓包程序吗，我的邮箱是 2080524998@qq.com

Taoyyp, 2024-03-20:
您好，我在运行captureAll.py时候，有时会出现”[fpga_udp]udp time out“这个问题，请问是怎么回事？

carrotlss, 2024-04-17:
> 您好，可以分享一下您的抓包程序吗，我的邮箱是 [2080524998@qq.com](mailto:2080524998@qq.com)

您好，也希望能分享一下，951173608@qq.com

zhoXR, 2024-04-23:
> 您好，可以分享一下您的抓包程序吗，我的邮箱是[2080524998@qq.com](mailto:2080524998@qq.com)

我用的就是作者的

zhoXR, 2024-04-23:
> > 您好，可以分享一下您的抓包程序吗，我的邮箱是[2080524998@qq.com](mailto:2080524998@qq.com)
> 
> 您好，也希望能分享一下，[951173608 @qq.com](mailto:951173608@qq.com)
我用的就是作者的源代码

zhoXR, 2024-04-23:
> 我没记错的话当时应该是这样设计的：
> 
> 1. 有一个udp数据包接收队列，在一个udp接收线程里面每次从socket接收固定字节数据包（dca1000每次会发固定字节的udp数据过来，如果雷达最后一帧数据不足这么多字节则会补零，该数据包的结构可以看官方手册）存入该队列。
> 2. 调用[udp_read_thread_get_frames](https://github.com/gaoweifan/pyRadar/blob/main/fpga_udp/src/main.cpp#L296)时会从该队列依次取出个数据包出来看数据包的序号，如果正好是一帧的起始所在的数据包，则从该包开始，取出指定帧数所需要的所有数据包出来，否则可以丢掉该包并检查下一个数据包的序列号。
> 
> 显然问题就在这里，如果要改的话，我建议在udp接收线程里，取到数据包后不要直接存入队列，而是取payload放入一个连续的图形，这个图形正好是一帧的大小的整倍，等前面的数据包填满这个灯光以后将一整帧存入队列，剩下多余的数据（下一帧的起始）存入新的灯光的起始。这样你就可以从队列里取出一整帧数据，而不会导致丢掉一部分下一帧索引的数据导致这帧无法使用被丢弃。

DoubleMiaoLi, 2024-07-14:
> 您好，我在运行captureAll.py时候，有时会出现”[fpga_udp]udp time out“这个问题，请问是怎么回事？

您好，请问这个问题解决了吗？

Taoyyp, 2024-07-17:
> > 您好，我在运行captureAll.py时候，有时会出现”[fpga_udp]udp time out“这个问题，请问是怎么回事？
> 
> 您好，请问这个问题解决了吗？

有时候程序端口占用了，所以会出现这种情况。具体我没有修改...

Ningoose, 2024-10-18:
> > 您好，我在运行captureAll.py时候，有时会出现”[fpga_udp]udp time out“这个问题，请问是怎么回事？
> 
> 您好，请问这个问题解决了吗？

您好，请问现在这个问你解决了吗？我好像并没有程序占用端口，还是始终显示[fpga_udp]udp time out