gaoweifan/pyRadar #19: AWR2243 Cascade SOP setting？

bella-jy, 2024-07-11:
您好，我在用AWR2243的级联雷达，到了烧录固件补丁这一步，出现了Device map 1: SOP 4 mode failed with error -1
我想是SOP配置的问题，请问应该怎么配置2243 cascade的SOP呀，我看您用的好像是2243的单片雷达，直接在板子上设置的。
但是cascade是在mmwave studio上设置的，卡在这个问题上了，求教！！感谢！！！

gaoweifan, 2024-07-11:
AWR2243确实是通过SPI设置的SOP，并不是直接通过板子上的开关控制的，具体请参考源码[mmw_example_nonos.cpp](https://github.com/gaoweifan/pyRadar/blob/f4673ad0e1bd9679ed2a4f4df3825b16c2a8e88a/fpga_udp/src/mmwaveDFP_2G/ti/example/mmWaveLink_SingleChip_NonOS_Example/mmw_example_nonos.cpp#L3705)。这个烧录固件的源码是TI官方提供的[mmWave-DFP-2G](https://www.ti.com/tool/MMWAVE-DFP)中的一部分，如果出现问题请咨询TI。

bella-jy, 2024-07-11:
感谢！原来是通过SPI设置的，AWR2243它默认是SOP4，我看您的教程上是SOP0，是不是要设置成SOP0呀？

gaoweifan, 2024-07-12:
我又看了一下[AWR2243的数据手册](https://www.ti.com/lit/gpn/awr2243)，板子上的开关是在启动的时候控制SOP(Sense On Power)的，启动后开关就不起作用了，可以通过SPI切换SOP模式。而且cascade板子也有SOP引脚，每个AWR2243的SOP都有引出，详情请参考[MMWCAS-RF-EVM用户手册](https://www.ti.com/lit/pdf/swru553)。启动时需要使SOP0引脚为高电平，即设置SOP为mode 4，亦即[SOP2 SOP1 SOP0] = [0 0 1] -> Functional SPI mode，具体启动过程请参考[启动流程文档](https://www.ti.com/lit/pdf/spracr5)。

bella-jy, 2024-08-20:
> 我又看了一下[AWR2243的数据手册](https://www.ti.com/lit/gpn/awr2243)，板子上的开关是在启动的时候控制SOP(Sense On [...]

麻烦请教一下如何通过SPI切换SOP模式呀？看了TI的文档一直没找到方法，万分感谢！🙏

gaoweifan, 2024-08-20:
首先上电时SOP0引脚就得为高电平，处于SPI工作模式，然后才能通过SPI控制SOP模式，控制代码可以参考[mmw_example_nonos.cpp](https://github.com/gaoweifan/pyRadar/blob/f4673ad0e1bd9679ed2a4f4df3825b16c2a8e88a/fpga_udp/src/mmwaveDFP_2G/ti/example/mmWaveLink_SingleChip_NonOS_Example/mmw_example_nonos.cpp#L3705)。