gaoweifan/pyRadar #20: 关于cfg文件的配置问题

Ningoose, 2024-08-23:
你好，我是毫米波雷达的初学者，我对从mmWave_Demo_Visualizer保存下来的cfg文件中有两个问题。
一是Scene Selection中的各个选项具体数值该怎么配置？在哪个文档里可以了解呢？
[image]

二是打开cfg文件，星号下边的各个参数后面的各个数值分辨代表什么意思呀？我对照mmwave studio好像也对应不上(比如profileCfg)，有些则完全没理解(比如channelCfg的15 7 0)。而且在mmWave_Demo_Visualizer中也没找到能调节这些参数的选项，只有Scene Selection。请问这些参数在哪里可以详细了解呀？

gaoweifan, 2024-08-23:
`%`开头的是注释，不会对芯片进行任何配置，只是把你在visualizer里设置的参数写了出来。这些参数是根据星号下方的配置值计算得到的，计算公式可以参考TI的[chirp编程手册](https://www.ti.com.cn/lit/an/swra553a/swra553a.pdf)。而这些配置项如channelCfg、profileCfg等含义则可以在[mmwave SDK](https://www.ti.com.cn/tool/cn/MMWAVE-SDK)用户手册中找到。

Ningoose, 2024-08-23:
感激不尽！