gaoweifan/pyRadar #21: 在CaptureAll获取点云数据出现magic words不匹配

kurci-silence, 2024-08-25:
你好，作者。我在使用您的CaptureALL程序获取Raw data和片内预处理的点云数据中遇到了magic words不匹配问题。
如图
[image]
received serial data bytes是有可观的数据量，但是结果显示magic word不匹配。
我在程序中调试发现每次匹配21436587这一组magic word时候，串口中的数据都无法正常匹配上
[image]
请问作者知道是大概哪里有问题或者有没有解决的思路呢？是不是与我雷达放置的位置有关，我放置在桌面距离我大概40~50cm斜对，高度在颈部的位置

gaoweifan, 2024-08-26:
点云数据是通过uart传输的，可能是传输过程出现了一些问题，请检查uart的波特率是否正确，USB线长度也不能太长。也有可能是固件做了修改，可以重新刷写官方固件试试。

kurci-silence, 2024-08-26:
非常感谢您的意见，现在问题已经得到解决，是波特率设置为115200导致的，现在我调为921600已经解决问题了。