gaoweifan/pyRadar #22: CaptureAll程序运行重复出现time out

kurci-silence, 2024-09-30:
作者好，之前在使用CaptureAll程序一切正常，后续由于系统崩溃更换到win10系统，我重新部署了Ti mmwave工作环境，并在mmwave studio上验证正常，重新配置好作者程序所需环境后，我运行出现了time out问题，暂时没有想到好的解决办法，怀疑可能是DCA板子数据链接出现了问题？
下边有三幅运行结果图，前两幅是time out出现问题，第三图是运行最终报错结果
[image]
[image]
[image]

kurci-silence, 2024-09-30:
不知道author有没有好的建议或者解决办法？非常感谢

gaoweifan, 2024-09-30:
建议检查网口ip地址有没有按要求设置，看起来像是网络传输问题