gaoweifan/pyRadar #24: captureall+1843的问题，无法获取采集后的bin文件

xwang612, 2024-10-10:
作者大大，你好，冒昧打扰，程序可以完整运行，在按下enter后，DCA1000指示灯闪烁，采集数据，这是控制台打印信息，start等号是我加的print，但是，在指示灯闪烁期间，程序就停止了，只执行配置命令和开始采集，后续所有的打印信息控制台也没有，也没有找到生成的bin文件和npy文件，这是什么地方出现问题了呢？感谢作者大大的解答
press ENTER to start capture...
data reading process started
Start streaming
>>> sensorStart
b'sensorStart\n'
===========start==================================
min Packet Num: 6752
PS C:\Users\wangxiaofeng>

xwang612, 2024-10-10:
我使用wireshark抓包可以看到确实向电脑发送了数据包
[image]

gaoweifan, 2024-10-10:
试试把fastRead_in_Cpp的参数sortInC=True改为False

haolin23, 2025-01-16:
请问您解决了吗，我也遇到这个问题，把sortInC=True改为False也是这样

DlwlrmaL, 2025-11-06:
我在AWR2243上也遇到了这个问题
press ENTER to start capture...
Start streaming
Async event: Frame trigger 

Sensor Start successful for deviceMap 1 

======================================================================

PS C:\Users\123456\Desktop\pyRadar-main>  希望作者能提供一些帮助，谢谢了