gaoweifan/pyRadar #27: `testGTrack.py` 文件无法正常处理文件中给出的样例数据

Maple-YZ, 2024-12-09:
### 问题描述

配置好环境后直接运行 `python testGTrack.py` 指令，运行过程看着正常，但最终输出的tNum变量是0，前面有报两行类似Warning的内容但因为之前没有专门系统学习过C++所以不太清楚具体含义，在C的原代码中也没有找到对应的函数/变量

想问一下大佬这个具体是什么原因呢？排查了很久但有点无从下手

### 复现

配置好环境后运行 `python testGTrack.py` 指令

### 运行截图

[image]
> use3D = True，这里加了一个print模块的语句用于确认C的相关模块被正常返回，未改动任何处理逻辑

[image]
> use3D = False，print语句同上

后续更改了程序代码并尝试接入我们自己的数据，也一样无法使用，同样表现为返回值的tNum=0

### 运行环境

系统： ubuntu22
python：3.10.15
cppyy库安装方式： conda install -c conda-forge cppyy
其余库均用pip install 安装当前默认版本

gaoweifan, 2024-12-09:
用低版本cppyy-2.4.0试试？3.x.x是大版本升级，改了很多东西，可能不兼容了
```shell
conda install conda-forge::cppyy==2.4.0
```

gaoweifan, 2024-12-10:
已修复最新版本cppyy-3.1.2中找不到符号的问题，请查看最新commit

closed 2024-12-10 by gaoweifan

Maple-YZ, 2024-12-10:
感谢！本来还在捣鼓旧版本的一些问题，换了新代码确实可以直接跑了

gaoweifan, 2024-12-10:
刚才改的比较匆忙，[mNum](https://github.com/gaoweifan/pyRadar/blob/f22aeaf44b368cb8771aafe019185bb9553e58ee/testGtrack.py#L131C5-L131C62)取值没改过来，不会随着输入点云数量变化，现在改过来了，见最新提交