gaoweifan/pyRadar #30: Windows配置fpga_up安装成功，在pip list和PyCharm的第三方库里都能看到，但是运行报错

robotswj, 2025-01-18:
您好，按照教程安装了包，但是fpga_udp似乎有点问题，没法成功加载，请帮忙看下是什么问题，万分感谢！

D:\anaconda\python.exe D:\工作及论文\2025\TI雷达开发\pyRadar-main\captureAll.py 
Traceback (most recent call last):
  File "D:\工作及论文\2025\TI雷达开发\pyRadar-main\captureAll.py", line 3, in <module>
    from mmwave.dataloader import DCA1000
  File "D:\工作及论文\2025\TI雷达开发\pyRadar-main\mmwave\__init__.py", line 13, in <module>
    from . import dataloader
  File "D:\工作及论文\2025\TI雷达开发\pyRadar-main\mmwave\dataloader\__init__.py", line 13, in <module>
    from .adc import *
  File "D:\工作及论文\2025\TI雷达开发\pyRadar-main\mmwave\dataloader\adc.py", line 22, in <module>
    import fpga_udp
ImportError: DLL load failed while importing fpga_udp: 找不到指定的模块。

gaoweifan, 2025-01-20:
确定fpga_udp模块按照要求安装了么，安装到什么环境下的，pip list看一下有没有，然后再看一下安装路径有没有dll文件生成

gaoweifan, 2025-01-20:
可以把你安装过程所有命令及其日志输出都贴上来么

robotswj, 2025-01-21:
好的，我这里整理下，谢谢~

gaoweifan, 2025-01-24:
你pycharm里用的是安装时选中的conda的base环境么

robotswj, 2025-01-25:
是的，您好，问题解决了，我看了下是系统里少一个fdti的dll，安装的时候有问题。目前跑起来了，但是没有采集数据。还需要您帮助下，下面是显示的log，最后是这个退出的Process finished with exit code -1073741819 (0xC0000005)。网口设置没问题，sop0-2设置是100

D:\anaconda\python.exe -X pycache_prefix=C:\Users\Administrator\AppData\Local\JetBrains\PyCharmCE2023.3\cpython-cache "D:/program/PyCharm Community Edition 2023.3.3/plugins/python-ce/helpers/pydev/pydevd.py" --multiprocess --qt-support=auto --client 127.0.0.1 --port 7159 --file D:\工作及论文\2025\TI雷达开发\pyRadar-main\captureAll.py 
Connected to pydev debugger (build 233.13763.11)
wait for reset
>>> sensorStop
b'sensorStop\nIgnored: Sensor is already stopped\n\rDone\n\rmmwDemo:/>'
>>> flushCfg
b'flushCfg\nDone\n\rmmwDemo:/>'
>>> dfeDataOutputMode 1
b'dfeDataOutputMode 1\nDone\n\rmmwDemo:/>'
>>> channelCfg 15 7 0
b'channelCfg 15 7 0\nDone\n\rmmwDemo:/>'
>>> adcCfg 2 1
b'adcCfg 2 1\nDone\n\rmmwDemo:/>'
>>> adcbufCfg -1 0 1 1 1
b'adcbufCfg -1 0 1 1 1\nDone\n\rmmwDemo:/>'
>>> profileCfg 0 77 974 7 40 0 0 100 1 64 2000 0 0 30
b'profileCfg 0 77 974 7 40 0 0 100 1 64 2000 0 0 30\nDone\n\rmmwDemo:/>'
>>> chirpCfg 0 0 0 0 0 0 0 1
b'chirpCfg 0 0 0 0 0 0 0 1\nDone\n\rmmwDemo:/>'
>>> chirpCfg 1 1 0 0 0 0 0 4
b'chirpCfg 1 1 0 0 0 0 0 4\nDone\n\rmmwDemo:/>'
>>> chirpCfg 2 2 0 0 0 0 0 2
b'chirpCfg 2 2 0 0 0 0 0 2\nDone\n\rmmwDemo:/>'
>>> frameCfg 0 2 32 0 200 1 0
b'frameCfg 0 2 32 0 200 1 0\nDone\n\rmmwDemo:/>'
>>> lowPower 0 0
b'lowPower 0 0\nDone\n\rmmwDemo:/>'
>>> guiMonitor -1 1 1 1 1 1 1
b'guiMonitor -1 1 1 1 1 1 1\nDone\n\rmmwDemo:/>'
>>> cfarCfg -1 0 2 8 4 3 0 15 1
b'cfarCfg -1 0 2 8 4 3 0 15 1\nDone\n\rmmwDemo:/>'
>>> cfarCfg -1 1 0 8 4 4 1 15 1
b'cfarCfg -1 1 0 8 4 4 1 15 1\nDone\n\rmmwDemo:/>'
>>> multiObjBeamForming -1 1 0.5
b'multiObjBeamForming -1 1 0.5\nDone\n\rmmwDemo:/>'
>>> clutterRemoval -1 0
b'clutterRemoval -1 0\nDone\n\rmmwDemo:/>'
>>> calibDcRangeSig -1 0 -5 8 256
b'calibDcRangeSig -1 0 -5 8 256\nDone\n\rmmwDemo:/>'
>>> extendedMaxVelocity -1 0
b'extendedMaxVelocity -1 0\nDone\n\rmmwDemo:/>'
>>> lvdsStreamCfg -1 0 1 0
b'lvdsStreamCfg -1 0 1 0\nDone\n\rmmwDemo:/>'
>>> compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0
b'compRangeBiasAndRxChanPhase 0.0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0 1 0\nDone\n\rmmwDemo:/>'
>>> measureRangeBiasAndRxChanPhase 0 1.5 0.2
b'measureRangeBiasAndRxChanPhase 0 1.5 0.2\nDone\n\rmmwDemo:/>'
>>> CQRxSatMonitor 0 3 4 99 0
b'CQRxSatMonitor 0 3 4 99 0\nDone\n\rmmwDemo:/>'
>>> CQSigImgMonitor 0 31 4
b'CQSigImgMonitor 0 31 4\nDone\n\rmmwDemo:/>'
>>> analogMonitor 0 0
b'analogMonitor 0 0\nDone\n\rmmwDemo:/>'
>>> aoaFovCfg -1 -90 90 -90 90
b'aoaFovCfg -1 -90 90 -90 90\nDone\n\rmmwDemo:/>'
>>> cfarFovCfg -1 0 0 2.40
b'cfarFovCfg -1 0 0 2.40\nDone\n\rmmwDemo:/>'
>>> cfarFovCfg -1 1 -0.32 0.32
b'cfarFovCfg -1 1 -0.32 0.32\nDone\n\rmmwDemo:/>'
>>> calibData 0 0 0
b'calibData 0 0 0\nDone\n\rmmwDemo:/>'
>>> frameCfg 0 2 32 10 200.0 1 0.0
b'frameCfg 0 2 32 10 200.0 1 0.0\nDone\n\rmmwDemo:/>'
676 1456
[WzSerialportPlus::open()]: open success.
data reading process created
{'startFreq': 77, 'idleTime': 974.0, 'adc_valid_start_time': 7.0, 'rampEndTime': 40.0, 'freq_slope': 100.0, 'txStartTime': 1.0, 'samples': 64, 'sample_rate': 2000, 'chirps': 32, 'frame_periodicity': 200.0, 'rx': 4, 'tx': 3, 'IQ': 2, 'bytes': 2}
{'platfrom': '18', 'rxAntMask': 15, 'txAntMask': 7, 'dataSize': 2, 'dataType': 1, 'chirpMode': 1, 'chirpStartIdx': 0, 'chirpEndIdx': 2, 'numSubframes': 1, 'numRxChan': 4, 'numTxChan': 3, 'datacard_dataLoggingMode': 'raw'}
LVDSDataSizePerChirp:1280 must <= maxSendBytesPerChirp:152100
System connection check: success
FPGA Version : 2.8 [Record]
Config fpga: success
Config record packet delay: success

Process finished with exit code -1073741819 (0xC0000005)

gaoweifan, 2025-01-25:
你先试试把dca采集原始数据相关代码都屏蔽掉，只留下uart采集点云数据，看看能不能运行，能的话可能是遇到了一些难以解释的兼容性问题，建议换Linux系统，或者直接用mmwave studio采集原始数据