gaoweifan/pyRadar #31: 使用captureAll.py和realTimeProc.py后无效果

Chenhongkun2001, 2025-02-13:
大佬您好！

我在VS Code中以调试模式运行captureAll.py时，终端结果仅会出现如下图所示的内容：

[image]

在摁下ENTER后，DCA1000EVM上DATA_TRANS_PROG_LED1并未闪烁（若使用mmwave studio，在实时采集期间，该LED会一直闪烁直到采集结束），随后终端**立即出现**如下内容，最终也并未形成.bin文件存储在PC上。

[image] 

若我直接运行captureAll.py，而非调试模式，则会出现如下结果，其中报出 FileNotFoundError: [Errno 2] No such file or directory: 'configFiles/xNR1843 profile 3D.cfg' 错误。

[image]

而在VS Code中以调试模式运行realTimeProc.py时，终端结果如下：

[image]

在摁下ENTER后，DCA1000EVM上DATA_TRANS_PROG_LED会**持续闪烁**，甚至直到结束调试模式、关闭VS Code。仅对DCA1000EVM断电后该闪烁才会消失。

若我直接运行realTimeProc.py，而非调试模式，则会出现与captureAll.py在运行模式下的相同结果。

我已经按照要求装好所有所需软件内容，硬件（AWR1843BOOST，DCA1000EVM）连接也按照要求接好。我使用的所有.cfg文件均是您main分支中的configFiles提供的，并将cf.json文件中的lvdsMode更改为2，lvdsStreamCfg的第三个参数需设置为1。代码除了端口号外均未进行修改。

Chenhongkun2001, 2025-02-19:
> 检查1843板子是否按要求刷写了xwr18xx_mmw_demo固件，cf配置文件为手动编制，其格式参考了DCA1000EVM CLI所使用的json配置文件，详情见[DCA1000EVM CLI Software User Guide](https://e2e.ti.com/cfs-file/__key/communityserver-discussions-components-files/1023/TI_5F00_DCA1000EVM_5F00_CLI_5F00_Software_5F00_UserGuide.pdf)。

您好，我参考了https://e2echina.ti.com/blogs_/b/the_process/posts/awr1843boost-mmw-demo 中写的使用Uniflash烧写的教程 同时在烧写结束后关闭uniflash工具软件。断开AWR1843BOOST电源，将AWR1843BOOST的SOP设置为SOP0=1（短接），其他为0。再执行完这一步的时候 接着执行您在Prerequisites/Hardware  硬件/or xWR1843  用于 xWR1843的硬件连接这部分。结果仍然是如上面所提到的那样，此外cf. json和cfg的内容均没有变化。  请问您这该如何解决。这是否还和其他因素有关？

gaoweifan, 2025-02-19:
看样子AWR1843BOOST的配置没什么问题，如果确定在mmwave studio内能正常捕获的话，问题有可能是出在了DCA1000EVM上，检查一下DCA1000EVM与AWR1843BOOST之间连接的60针HD排线是否松动，DCA1000EVM拨码开关是否设置正确，SW2.1是否为LVDS_CAPTURE，SW2.2是否为ETH_STREAM，SW2.3是否为AR1642_MODE，SW2.4是否为RAW_MODE。若确定DCA1000EVM也无问题的情况下，还是无法正常采集原始数据，则有可能遇到了兼容性问题，要么考虑更换Linux操作系统，要么考虑使用采集方法一，见[issue 2](https://github.com/gaoweifan/pyRadar/issues/2#issuecomment-1564261036)