gaoweifan/pyRadar #34: 关于帧间不连续的询问

haolin23, 2025-03-11:
作者您好，我使用代码时把numframes设置为2，发现每一帧内chirp的相位都是连续的，但是相邻两帧之间的相位存在不连续。

不知道这是不是代码的正常现象，以及如果我想要实现相邻两帧的相位连续的话，有没有修改代码的可能建议。

非常感谢您的回复和解答。

gaoweifan, 2025-03-11:
这是正常的，帧与帧之间是有间隔时间的，具体请参考ti的手册，比如[这里](https://www.ti.com/lit/an/swra553a/swra553a.pdf)就提到了**inter frame time**，且其间隔时间远大于相邻chirp，相位肯定不是连续的

haolin23, 2025-03-13:
感谢解答。

我还有一个疑问：为了增加一个frame的采集时长，我在尝试增加配置文件里frameCfg中的number of loops值，发现当number of loops设置为96时无法运行（报错如下），但是当缩小到比如64时就可以正常运行，请问一下这可能是什么原因导致的。

（按照MMWAVE SDK User Guide，number of loops的范围应该在1-255，没有超出。另外我在testParam.ipynb测试了cfg文件，也没有报错信息。）

Traceback (most recent call last):
  File "E:\pyRadar\pyRadar\realTimeProc_my.py", line 264, in <module>
    data_buf = dca.fastRead_in_Cpp_thread_get(numframes,verbose=True,sortInC=True) # 方法一（推荐）
  File "E:\pyRadar\pyRadar\mmwave\dataloader\adc.py", line 828, in fastRead_in_Cpp_thread_get
    print("received packet num:%d,expected packet num:%d,loss:%.2f%%"%(receivedPacketNum,expectedPacketNum,(expectedPacketNum-receivedPacketNum)/expectedPacketNum*100))

ZeroDivisionError: division by zero

gaoweifan, 2025-03-14:
由于要在片上实时处理数据，数据量不能太大，不然处理不过来，具体限制得看sdk demo怎么写的了，testParam测试的仅仅是根据硬件特性来确定参数的合理性，没有考虑片上处理软件的限制，如果不用片上处理，单独用mmwave studio的固件采集，可能就能在大数据量的情况下正常运行