gaoweifan/pyRadar #38: 采集.bin数据包过大问题

kurci-silence, 2025-05-13:
你好，up，我在用captureall.py程序采集.bin数据时出现数据丢失问题。由于我需要提高慢时间上采样率，因此我在.cfg配置中的帧率设置非常高，帧时间为1ms，因此我采集1min的.bin数据包大概在2.7G左右，就会出现数据包中后半段数据丢失问题，我尝试用winhex常看内容，发现都是0，在.py运行结果上也出现了丢包提示。

[image]
我之前有注意到ti自带脚本抓取数据一次最多只能保存1g的bin文件，多出的数据会以多个bin文件分开保存，不知道您觉得这两点是否相同情况？
另外，我将采集时间减半，此时bin数据包为1.3G左右，经过解析后的数据是正常的，没有出现丢失状况，上述情况您认为什么原因导致的？如果您能答疑我将非常感激

kurci-silence, 2025-05-13:
另外，我修改了.json配置内容，将最大保存改为4096Mb，仍是类似的问题

[image]

gaoweifan, 2025-05-13:
.json设置了也没用，因为我只解析了DCA1000Config里的内容：
https://github.com/gaoweifan/pyRadar/blob/849ff652301dd15e5f33fb048c455b386b329203/mmwave/dataloader/adc.py#L967-L972
丢包更像是因为DCA1000板载内存满了导致的，如果数据量过大，超过了网口能传输的最大带宽，就会出现数据积压在板载DDR内直至被新数据覆盖的情况，可以运行[testParam.ipynb](/gaoweifan/pyRadar/blob/main/testParam.ipynb)看看FPGA throughput是否满足约束条件。

kurci-silence, 2025-05-14:
> .json设置了也没用，因为我只解析了DCA1000Config里的内容：
> 
> [pyRadar/mmwave/dataloader/adc.py](https://github.com/gaoweifan/pyRadar/blob/849ff652301dd15e5f33fb048c455b386b329203/mmwave/dataloader/adc.py#L967-L972)
> 
> Lines 967 to 972 in [849ff65](/gaoweifan/pyRadar/commit/849ff652301dd15e5f33fb048c455b386b329203)
> 
>  payload = (ConfigLogMode[json_data['DCA1000Config']['dataLoggingMode']]+ 
>      ConfigLvdsMode[json_data['DCA1000Config']['lvdsMode']]+ 
>      ConfigTransferMode[json_data['DCA1000Config']['dataTransferMode']]+ 
>      ConfigCaptureMode[json_data['DCA1000Config']['dataCaptureMode']]+ 
>      ConfigFormatMode[json_data['DCA1000Config']['dataFormatMode']]+ 
>      ConfigTimer) 
> 
> 丢包更像是因为DCA1000板载内存满了导致的，如果数据量过大，超过了网口能传输的最大带宽，就会出现数据积压在板载DDR内直至被新数据覆盖的情况，可以运行[testParam.ipynb](/gaoweifan/pyRadar/blob/main/testParam.ipynb)看看FPGA throughput是否满足约束条件。