gaoweifan/pyRadar #43: 脚本因UDP超时和ZeroDivisionError而失败

wushuanglang, 2025-08-04:
您好，我在Jetson Xavier NX (Ubuntu 18.04)上运行 captureAll.py 连接DCA1000时遇到问题，脚本因UDP超时和ZeroDivisionError而失败。
可以工作的情况：
同样的DCA1000和iwr1843boost，在Windows电脑上使用mmWave Studio可以完美工作。使用mmWave_Demo_Visualizer 3.6.0也可以正常工作，之后我使用uniflash烧录了xwr18xx_mmw_demo.bin，去jetson下开发。
我在Jetson xavier nx上做的调试工作如下：

静态IP正确：我的 eth0 网卡已正确配置静态IP 192.168.33.30。如下图所示：
[image]
防火墙开放：ufw 未安装，iptables -L 显示INPUT链的默认策略为ACCEPT。
[image]
权限和环境正确：我使用了 sudo 和虚拟环境的完整路径来运行脚本。运行结果如下：
[image]
Wireshark的关键发现：当运行脚本时，Wireshark显示没有任何UDP包从Jetson发送出去。这证明脚本甚至没有尝试通过网络发送FPGA配置指令。
[image]
[image]
看起来脚本内部存在某个问题，在我的系统上阻止了它初始化网络通信。希望能得到您的帮助。
这个是昨天上传的，不小心关掉了，再上传一遍。

wushuanglang, 2025-08-04:
我换了一快开发板，现在报错貌似是雷达启动失败，下面是运行captureAll.py的截图：

[image]
下面是运行realTimeProc.py的截图：

[image]

wushuanglang, 2025-08-05:
> 我换了一快开发板，现在报错貌似是雷达启动失败，下面是运行captureAll.py的截图：
> 
> <img alt="Image" width="2000" height="1043" src="https://private-user-images.githubusercontent.com/145752708/473959809-534a8d29-c850-4b0e-825a-62856fedac2a.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ0MDk2MjUsIm5iZiI6MTc1NDQwOTMyNSwicGF0aCI6Ii8xNDU3NTI3MDgvNDczOTU5ODA5LTUzNGE4ZDI5LWM4NTAtNGIwZS04MjVhLTYyODU2ZmVkYWMyYS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwNVQxNTU1MjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02MzU1NTMyZGNjZGE3YjJiOWZkMDU0Mjg0MzJiOWViY2I2NDYyZGU5ZDA2YzZhNGRkMWM1N2ZmODhhZmY0YzY0JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.gfSSAiHqYsGGakwlEmYsULHdvC1cBMMnQVPwfxIIhhg"> 下面是运行realTimeProc.py的截图： <img alt="Image" width="2000" height="965" src="https://private-user-images.githubusercontent.com/145752708/473959956-e61f0d81-c053-4d88-b4de-7d525c9f4494.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3NTQ0MDk2MjUsIm5iZiI6MTc1NDQwOTMyNSwicGF0aCI6Ii8xNDU3NTI3MDgvNDczOTU5OTU2LWU2MWYwZDgxLWMwNTMtNGQ4OC1iNGRlLTdkNTI1YzlmNDQ5NC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjUwODA1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI1MDgwNVQxNTU1MjVaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0xZWNhNmI3NzdlYjZkMGE3MTc5ZWJhNTcxZjkyMGQ2YTcwZWEyY2U2NTU2NWI3MzgwNjJhZDRmZTQwNDE1NTgxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCJ9.yO3wLIIolfexSr_1aSgu2BL05Ck5sXqVLzxY-dIO1yQ">

我问题已经解决了，我使用脚本和minicom调试了一下，找到了问题所在，[udp] time out的原因是串口出了问题，没有正确发送数据，导致雷达没配置成功，雷达没启动，当然不会接受到数据，于是就导致了[udp] time out，最后导致了一个除零错误。我在香橙派5pro上已成功解决问题，获取到了点云数据。
但是在jetson xavier nx上遇到了以太网传输不了数据的问题，初步分析是nx被设置了固定ip192.168.1.160，我甚至无法通过netplan和NetworkManager对其做出修改，改了也不生效。而DCA1000要求的网段是33，我试过使用这个命令添加192.168.33.30这个ip：sudo ip addr add 192.168.33.30/24 dev eth0，但仍然重置雷达和fpga时超时，无法通信。
目前问题正在解决中，如果你知道解决思路的话，欢迎指教。

gaoweifan, 2025-08-12:
我记得我以前遇到过类似的问题，大概是4年前我也在Jetson Xavier NX上连接DCA1000EVM采集过数据，我找了一下当年的文件，当时我把配置过程记录到了`Jetson与DCA1000EVM相连的网络配置.txt`中，该文件内容如下文所示，希望对你有所帮助。
```txt
运行下列命令，进入网络连接设置界面（需要xserver）
sudo nm-connection-editor
左下角加号添加Ethernet，名称任意（如：DCA1000EVM），IPV4改为manual，添加一个IP地址192.168.33.30，子网掩码255.255.255.0
保存后运行下列命令启用连接
sudo nmcli c up DCA1000EVM
查看是否连接上
sudo nmcli d

打开转发开关
要让iptables的端口转发生效，首先需要打开转发开关
方法一：临时打开，重启后失效
$sudo su
#echo 1 >/proc/sys/net/ipv4/ip_forward

方法二：永久打开，重启依然有效
编辑/etc/sysctl.conf文件，将net.ipv4.ip_forward=1前面的#注释去掉，保存文件，然后执行sudo sysctl -p使其生效

设置端口转发，能够通过WiFi所在局域网在其他设备上采集DCA板子数据
sudo iptables -P INPUT ACCEPT
sudo iptables -P FORWARD ACCEPT
sudo iptables -P OUTPUT ACCEPT
sudo iptables -t nat -A PREROUTING -d 192.168.31.115 -p udp --dport 4096 -j DNAT --to-destination 192.168.33.180:4096
sudo iptables -t nat -A POSTROUTING -d 192.168.33.180 -p udp --dport 4096 -j SNAT --to 192.168.31.115

sudo iptables -t nat -A PREROUTING -d 192.168.31.115 -p udp --dport 4098 -j DNAT --to-destination 192.168.33.180:4098
sudo iptables -t nat -A POSTROUTING -d 192.168.33.180 -p udp --dport 4098 -j SNAT --to 192.168.31.115
```

wushuanglang, 2025-08-12:
> 我记得我以前遇到过类似的问题，大概是4年前我也在Jetson Xavier NX上连接DCA1000EVM采集过数据，我找了一下当年的文件，当时我把配置过程记录到了`Jetson与DCA1000EVM相连的网络配置.txt`中，该文件内容如下文所示，希望对你有所帮助。
> 
> 运行下列命令，进入网络连接设置界面（需要xserver）
> sudo nm-connection-editor
> 左下角加号添加Ethernet，名称任意（如：DCA1000EVM），IPV4改为manual，添加一个IP地址192.168.33.30，子网掩码255.255.255.0
> 保存后运行下列命令启用连接
> sudo nmcli c up DCA1000EVM
> 查看是否连接上
> sudo nmcli d
> 
> 
> 
> 
> 打开转发开关
> 要让iptables的端口转发生效，首先需要打开转发开关
> 方法一：临时打开，重启后失效
> $sudo su
> #echo 1 >/proc/sys/net/ipv4/ip_forward
> 
> 方法二：永久打开，重启依然有效
> 编辑/etc/sysctl.conf文件，将net.ipv4.ip_forward=1前面的#注释去掉，保存文件，然后执行sudo sysctl -p使其生效
> 
> 设置端口转发，能够通过WiFi所在局域网在其他设备上采集DCA板子数据
> sudo iptables -P INPUT ACCEPT
> sudo iptables -P FORWARD ACCEPT
> sudo iptables -P OUTPUT ACCEPT
> sudo iptables -t nat -A PREROUTING -d 192.168.31.115 -p udp --dport 4096 -j DNAT --to-destination 192.168.33.180:4096
> sudo iptables -t nat -A POSTROUTING -d 192.168.33.180 -p udp --dport 4096 -j SNAT --to 192.168.31.115
> 
> sudo iptables -t nat -A PREROUTING -d 192.168.31.115 -p udp --dport 4098 -j DNAT --to-destination 192.168.33.180:4098
> sudo iptables -t nat -A POSTROUTING -d 192.168.33.180 -p udp --dport 4098 -j SNAT --to 192.168.31.115

好的，非常感谢您的回复，我目前已经解决了问题，我的jetson xavier nx是定制的，其不支持千兆，但是DCA1000应该是被硬编码为了千兆，所以一直通信不上，后来我尝试使用交换机来充当中介，将千兆的数据转为百兆，可以在jetson xavier nx中成功获取到数据了。但是这种方法是权宜之计，数据传输速率不能超过百兆，我已经考虑更换开发板了。

Nick-Sun1018, 2025-11-20:
看了您们以前的回答，我是在Linux PC上操作的，之前能够成功采集数据，一次电脑同时连接了一个激光雷达和毫米波雷达之后，后面采集数据就是[udp] time out这个报错，单独使用1843能够在ros下获得点云，试了您之前的解决方案，没有解决，请问还需要怎么排查？